mp3guy/ElasticFusion #131: PLY Export?

KoSv, 2018-01-20:
Hey! Great work, Thanks!
How can I export a poincloud or PLY?
greetz!

elliestath, 2018-01-22:
@KoSv if you press save then you should see a file with the extension .ply in your working directory, i.e. where you .klg file is.

edith-langer, 2018-02-09:
Do you know a tool to visualize the reconstruction when saving it as ply? I know that meshlab can do that, but the surfels are not rendered with their radius, but just as points. I would really like to see the surfels with the radius.
Any ideas?

KoSv, 2018-02-10:
perhaps: blender -> import ply -> particles -> use the particle color from "Col" in the vertex color tab?

edith-langer, 2018-02-13:
@KoSv thanks for your answer! I'm not a blender expert. What you suggest is a fixed defined size for the radius? Or could I assign the real radius values of each surfel to a particle?

KoSv, 2018-02-13:
@edith-langer what do you mean with "surfel" - > a vertex in the pointcloud?

edith-langer, 2018-02-14:
@KoSv Yes, with surfel I mean a vertex in the reconstruction (having xyz-coordinates, color, confidence, radius, ...)

KoSv, 2018-02-14:
@edith-langer ok... I din't know they have radius. does PLY support it?
if so, take a look in the "animation nodes" plugin for blender. you could grab the data ( if present as information in the pointcloud) and visualize it.
btw. blender has a python API. so anything custom is allowed! ;)
good luck!

wine3603, 2018-04-28:
@edith-langer Hi, if you have PCL installed, you can use pcl_ply2pcd to make it to be pcd files and you can get the xyzrgb infos. for "surfel", since it is not a common format or unit, you have to go inside the codes and export it. 
BTW ply files can be visualized by CloudCompare and Meshlab.

StoneCh168, 2018-11-27:
@edith-langer Try open3D! :D

closed 2022-05-27 by mp3guy