mp3guy/ElasticFusion #135: Elastic Fusion crashes almost every time when 'exit' is pressed

elliestath, 2018-01-30:
Hi,
I run ElasticFusion on my laptop (Dell inspiron 15 7000, GTX 1050 Ti, running on Ubuntu 16.04) and everything runs fine, the reconstruction is done, I save the 'ply' file but every time I press the 'exit' button, my whole system crashes, the mouse seems to move but not responding, the only solution is to force restart the system.
I had the same problem on a similar laptop. It'a been a month now and the system crashes every time, do you have any solution?

stdmkmehp, 2018-03-03:
I got the same problem and have no idea how to fix it. I tried an early version but it didn't work.
And I run ElasticFusion on my laptop(Intel Xeon E3-1505M + Quadro M1000M on ubuntu14.04).
Could someone help me, please?

chaowang15, 2018-03-11:
Got same problem with a laptop GTX 1070. Seems to be the driver problem. Anyone got the solution?

martinakos, 2018-03-29:
Same problem with a Dell Precision 7510 +  Intel® Xeon(R) CPU E3-1545M + Quadro M2000M on Ubuntu 16.04. Any solutions?

akatrevorjay, 2018-08-11:
Lol, same here, bahahaha

seamanj, 2019-05-05:
same here, every time I have to reboot my computer, it's so annoying?

minihops123, 2019-05-28:
I have the same problem. At linux you can kill ElasticFusion by pressing CTRL+ALT+ALTGR+\
Then you don't have to reboot every time

Quitino, 2020-03-29:
laptap.same problem.

closed 2022-05-27 by mp3guy