mp3guy/ElasticFusion #139: In real-time xtion, some questions arise, I hope someone can answer.Thanks

root29, 2018-02-23:
First of all, thank Thomas for his contribution.
When I ran in real time, I found that if the camera is a wall, the algorithm will lose the current camera pose, resulting in the point cloud can not be correctly matched, the entire three-dimensional map will deviate a lot. Whether the algorithm, for a single scene (such as pure walls, pure ground) algorithm in the camera pose estimation still need to be improved? I am not sure whether the equipment problem or algorithm, I hope to get your reply, thank you!

hanxiaoTan, 2018-04-09:
I have the same problem,because of it when matching a whole room I turn the sensor around,the camera pose will be rotated strangely.Does anybody have the solution of it ? thanks a lot

ssggle, 2020-06-22:
I also encountered the same problem, is there any solution?

ssggle, 2020-06-22:
I also encountered the same problem, is there any solution?@hanxiaoTan@root29 ，thanks

closed 2022-05-27 by mp3guy