mp3guy/ElasticFusion #145: Can I get a mesh using Elastic Fusion

RajathSwaroop, 2018-04-17:
Hi, I tried using Elastic Fusion with intel realsense D415 and it works great. However, it generates the point cloud map of the scene I record. I wish to get a mesh instead of point cloud. Is that possible??

RajathSwaroop, 2018-04-18:
For a better loop closure should I have a lower inlier threshold or greater? Also can you explain how to covariance threshold works for loop closure (-cv).

zc102241, 2018-07-20:
@RajathSwaroop i've also tried using ElasticFusion with intel realsense D435. I'm in trouble with "dev.set_frame_callback". There is no more such callback functions in librealsense2.
I tried to use the lambda function of rs2::sensor.run() ,but it failed. Could  u pls show some details of  ur Realsense2Interface.cpp file or give some hints about that?
Thanks  in advance

zc102241, 2018-07-20:
solved. the problem is something about the stream profiles  of sensors : )

LittleLampChen, 2018-09-18:
You can reconstruct a mesh by meshlab. 
however, the mesh is very noisy.

lixz123007, 2020-05-16:
Do you get a mesh you want? I use meshlab to render-shaders-splatting to get something really like mesh.But i cannot save as mesh! There is no face. Any way to make a real mesh?

densechen, 2020-05-16:
> @lixz123007 
You shoud export mesh in meshlab.

lixz123007, 2020-05-16:
> > @lixz123007
> > You shoud export mesh in meshlab.

How to export mesh?After i splatting,it is no face.Or your meaning is using other reconstruct way (like Possion?）then export.

densechen, 2020-05-16:
yes

lixz123007, 2020-05-19:
> yes

well，i thought elasticfusion like BundleFusion can output mesh ...
But thanks!

AN-1008, 2021-04-25:
I am wondering if the souce code of meshlab can be used.Then i can run the code of meshing in the terminal.

closed 2022-05-27 by mp3guy