mp3guy/ElasticFusion #153: Zed Stereolabs camera

rnunziata, 2018-07-12:
I wrote a Zed Stereolabs camera interface for ElasticFusion....if you think it would be good addition please let me know how send to you... not familiar with github software management for uploading.

iglew, 2018-07-25:
Hi @rnunziata, I would be interested in trying this out if you're offering, please? Is it possible to just put it in its own git repo and post it here please?

qaisarehman, 2019-10-15:
Hi, I know its an old post! but when ever you came here again! When im building the GUI, im facing this error. Please I will appreciate any feed back! thanks 

 `/home/tll/Downloads/ElasticFusion/ZEDGUI/src/MainController.cpp:402:83: error: no matching function for call to ‘GlobalModel::renderPointCloud(pangolin::OpenGlMatrix, const float&, const bool&, const bool&, const bool&, const bool&, const bool&, const bool&, const bool&, const int&, const int&)’
                                                       eFusion->getTimeDelta());
`

qaisarehman, 2019-11-01:
> Hi, I know its an old post! but when ever you came here again! When im building the GUI, im facing this error. Please I will appreciate any feed back! thanks [...]

There is a pre build executable in you archive. I managed to local all the dependencies. but now im stuck on 

Creating live capture... failed!
DeviceOpen using default: no devices found
Creating live capture... ./ElasticFusion: symbol lookup error: ./ElasticFusion: undefined symbol: _ZN2sl9InputTypeC1Ev

lets see if i can fix this! :)

Ryushane, 2020-05-19:
> > Hi, I know its an old post! but when ever you came here again! When im building the GUI, im facing this error. Please I will appreciate any feed back! thanks
> > `/home/tll/Downloads/ElasticFusion/ZEDGUI/src/MainController.cpp:402:83: error: no matching function for call to ‘GlobalModel::renderPointCloud(pangolin::OpenGlMatrix, const float&, const bool&, const bool&, const bool&, const bool&, const bool&, const bool&, const bool&, const int&, const int&)’ eFusion->getTimeDelta()); `
> 
> There is a pre build executable in you archive. I managed to local all the dependencies. but now im stuck on
> 
> Creating live capture... failed!
> DeviceOpen using default: no devices found
> Creating live capture... ./ElasticFusion: symbol lookup error: ./ElasticFusion: undefined symbol: _ZN2sl9InputTypeC1Ev
> 
> lets see if i can fix this! :)
> 
> update 01:
> 
> passed symbol look up error! it was related to the zed SDK i was using. figured i need 2.5 sdk, now stuck on another symbol error, related to pangolin.

Faced the same problem, have you solved it?

qaisarehman, 2020-05-21:
> Faced the same problem, have you solved it?

No I didnt solved the issue, i filed to rebuild the exe and unable to run it pre build either. I dropped it! 
I used the zed camera video and after a painful manual process, I was able to build a klg dataset, but the result was not good compare to the kinect when you use it with Elasticfusion.

akeaveny, 2020-09-10:
Hi, I'm curently trying to build with the zed GUI on Ubuntu 18, Cuda 10 & ZED SDK 3.2.

I'm running into problems with importing `#include <sl/Camera.hpp>` as my error starts with:
`/home/akeaveny/catkin_ws/src/ElasticFusion/GUI/src/Tools/ZedInterface.cpp: In constructor ‘ZedInterface::ZedInterface(int, int)’:
/home/akeaveny/catkin_ws/src/ElasticFusion/GUI/src/Tools/ZedInterface.cpp:33:18: error: ‘SUCCESS’ is not a member of ‘sl’
   if (err != sl::SUCCESS) {`

I also face this error in the MainController
`/home/akeaveny/catkin_ws/src/ElasticFusion/GUI/src/MainController.cpp: In member function ‘void MainController::run()’:
/home/akeaveny/catkin_ws/src/ElasticFusion/GUI/src/MainController.cpp:402:83: error: no matching function for call to ‘GlobalModel::renderPointCloud(pangolin::OpenGlMatrix, const float&, const bool&, const bool&, const bool&, const bool&, const bool&, const bool&, const bool&, const int&, const int&)’
                                                            eFusion->getTimeDelta());
`

Any help would be greatly appreciated @rnunziata, I know it's been a while.

sco09, 2021-04-01:
> No I didnt solved the issue, i filed to rebuild the exe and unable to run it pre build either. I dropped it!
> I used the zed camera video and after a painful manual process, I was able to build a klg [...]

I also build a klg file by using Zed2, however, when I use ElasticFusion to open this klg file, it stuck, the GUI is totally dark, do you know how is it going? Maybe the resolution is not correct? The resolution I used is 3480*1080. I saw the resolution in ElasticFusion is 640*480. What's the resolution you using when you build the klg by using Zed? Thanks!

qaisarehman, 2021-04-02:
its a painful procedure. Kinect v1 uses 640x480 resolution, and color depth for RGB and depth image is different! so first you have to rescale your RGB and depth images to 640x480 res, and then on Photoshop you have to check the color depth for both! thn you will be able to build a klg file using PNG2KLG. But in my experience the results were not good!

closed 2022-05-27 by mp3guy