mp3guy/ElasticFusion #154: Cmake core on JetsonTX2 Unable to find SuiteSparse

ZGX010, 2018-07-25:
Hi:
when i CMake the core find a Error that 
> CMake Error at FindSuiteSparse.cmake:115 (MESSAGE):
  Unable to find SuiteSparse
> Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)
but i have use apt-get install libsuitespare-dev to installed it.
And i use the jetsonTX2.can you hellp me?

ZGX010, 2018-07-25:
and when i dpkg -l | grep suitesparse it will print:  

>  ii  libsuitesparse-dev:arm64  1:4.4.6-1  arm64 libraries for sparse matrices computations (development files)
ii  libsuitesparseconfig4.4.6:arm64   1:4.4.6-1   arm64 configuration routines for all SuiteSparse modules

zssjh, 2018-07-26:
replace FindSuiteSparse.cmake with the file below!

MACRO(SUITESPARSE_REPORT_NOT_FOUND REASON_MSG)
  UNSET(SUITESPARSE_FOUND)
  UNSET(SUITESPARSE_INCLUDE_DIRS)
  UNSET(SUITESPARSE_LIBRARIES)
  UNSET(SUITESPARSE_VERSION)
  UNSET(SUITESPARSE_MAIN_VERSION)
  UNSET(SUITESPARSE_SUB_VERSION)
  UNSET(SUITESPARSE_SUBSUB_VERSION)
  # Do NOT unset SUITESPARSE_FOUND_REQUIRED_VARS here, as it is used by
  # FindPackageHandleStandardArgs() to generate the automatic error message on
  # failure which highlights which components are missing.

  # Note <package>_FIND_[REQUIRED/QUIETLY] variables defined by FindPackage()
  # use the camelcase library name, not uppercase.
  IF (SuiteSparse_FIND_QUIETLY)
    MESSAGE(STATUS "Failed to find SuiteSparse - " ${REASON_MSG} ${ARGN})
  ELSEIF (SuiteSparse_FIND_REQUIRED)
    MESSAGE(FATAL_ERROR "Failed to find SuiteSparse - " ${REASON_MSG} ${ARGN})
  ELSE()
    # Neither QUIETLY nor REQUIRED, use no priority which emits a message
    # but continues configuration and allows generation.
    MESSAGE("-- Failed to find SuiteSparse - " ${REASON_MSG} ${ARGN})
  ENDIF (SuiteSparse_FIND_QUIETLY)

  # Do not call RETURN(), s/t we keep processing if not called with REQUIRED.
ENDMACRO(SUITESPARSE_REPORT_NOT_FOUND)

# Specify search directories for include files and libraries (this is the union
# of the search directories for all OSs).  Search user-specified hint
# directories first if supplied, and search user-installed locations first
# so that we prefer user installs to system installs where both exist.
LIST(APPEND SUITESPARSE_CHECK_INCLUDE_DIRS
  ${SUITESPARSE_INCLUDE_DIR_HINTS}
  /opt/local/include
  /opt/local/include/ufsparse # Mac OS X
  /usr/local/homebrew/include # Mac OS X
  /usr/local/include
  /usr/local/include/suitesparse
  /usr/include/suitesparse # Ubuntu
  /usr/include)
LIST(APPEND SUITESPARSE_CHECK_LIBRARY_DIRS
  ${SUITESPARSE_LIBRARY_DIR_HINTS}
  /opt/local/lib
  /opt/local/lib/ufsparse # Mac OS X
  /usr/local/homebrew/lib # Mac OS X
  /usr/local/lib
  /usr/local/lib/suitesparse
  /usr/lib/suitesparse # Ubuntu
  /usr/lib)

# Given the number of components of SuiteSparse, and to ensure that the
# automatic failure message generated by FindPackageHandleStandardArgs()
# when not all required components are found is helpful, we maintain a list
# of all variables that must be defined for SuiteSparse to be considered found.
UNSET(SUITESPARSE_FOUND_REQUIRED_VARS)

# BLAS.
FIND_PACKAGE(BLAS QUIET)
IF (NOT BLAS_FOUND)
  SUITESPARSE_REPORT_NOT_FOUND(
    "Did not find BLAS library (required for SuiteSparse).")
ENDIF (NOT BLAS_FOUND)
LIST(APPEND SUITESPARSE_FOUND_REQUIRED_VARS BLAS_FOUND)

# LAPACK.
FIND_PACKAGE(LAPACK QUIET)
IF (NOT LAPACK_FOUND)
  SUITESPARSE_REPORT_NOT_FOUND(
[... 435 lines cut ...]
      SET(SUITESPARSE_IS_BROKEN_SHARED_LINKING_UBUNTU_SYSTEM_VERSION TRUE)
      MESSAGE(STATUS "Found system install of SuiteSparse "
        "${SUITESPARSE_VERSION} running on Ubuntu, which has a known bug "
        "preventing linking of shared libraries (static linking unaffected).")
    ENDIF (LSB_DISTRIBUTOR_ID MATCHES "Ubuntu" AND
      SUITESPARSE_LIBRARIES MATCHES "/usr/lib/libamd")
  ENDIF (LSB_RELEASE_EXECUTABLE)
ENDIF (CMAKE_SYSTEM_NAME MATCHES "Linux" AND
  SUITESPARSE_VERSION VERSION_EQUAL 3.4.0)

# Handle REQUIRED and QUIET arguments to FIND_PACKAGE
INCLUDE(FindPackageHandleStandardArgs)
IF (SUITESPARSE_FOUND)
  FIND_PACKAGE_HANDLE_STANDARD_ARGS(SuiteSparse
    REQUIRED_VARS ${SUITESPARSE_FOUND_REQUIRED_VARS}
    VERSION_VAR SUITESPARSE_VERSION
    FAIL_MESSAGE "Failed to find some/all required components of SuiteSparse.")
ELSE (SUITESPARSE_FOUND)
  # Do not pass VERSION_VAR to FindPackageHandleStandardArgs() if we failed to
  # find SuiteSparse to avoid a confusing autogenerated failure message
  # that states 'not found (missing: FOO) (found version: x.y.z)'.
  FIND_PACKAGE_HANDLE_STANDARD_ARGS(SuiteSparse
    REQUIRED_VARS ${SUITESPARSE_FOUND_REQUIRED_VARS}
    FAIL_MESSAGE "Failed to find some/all required components of SuiteSparse.")
ENDIF (SUITESPARSE_FOUND)

zssjh, 2018-07-26:
replace FindSuiteSparse.cmake with the code below

`MACRO(SUITESPARSE_REPORT_NOT_FOUND REASON_MSG)
  UNSET(SUITESPARSE_FOUND)
  UNSET(SUITESPARSE_INCLUDE_DIRS)
  UNSET(SUITESPARSE_LIBRARIES)
  UNSET(SUITESPARSE_VERSION)
  UNSET(SUITESPARSE_MAIN_VERSION)
  UNSET(SUITESPARSE_SUB_VERSION)
  UNSET(SUITESPARSE_SUBSUB_VERSION)
  # Do NOT unset SUITESPARSE_FOUND_REQUIRED_VARS here, as it is used by
  # FindPackageHandleStandardArgs() to generate the automatic error message on
  # failure which highlights which components are missing.

  # Note <package>_FIND_[REQUIRED/QUIETLY] variables defined by FindPackage()
  # use the camelcase library name, not uppercase.
  IF (SuiteSparse_FIND_QUIETLY)
    MESSAGE(STATUS "Failed to find SuiteSparse - " ${REASON_MSG} ${ARGN})
  ELSEIF (SuiteSparse_FIND_REQUIRED)
    MESSAGE(FATAL_ERROR "Failed to find SuiteSparse - " ${REASON_MSG} ${ARGN})
  ELSE()
    # Neither QUIETLY nor REQUIRED, use no priority which emits a message
    # but continues configuration and allows generation.
    MESSAGE("-- Failed to find SuiteSparse - " ${REASON_MSG} ${ARGN})
  ENDIF (SuiteSparse_FIND_QUIETLY)

  # Do not call RETURN(), s/t we keep processing if not called with REQUIRED.
ENDMACRO(SUITESPARSE_REPORT_NOT_FOUND)

# Specify search directories for include files and libraries (this is the union
# of the search directories for all OSs).  Search user-specified hint
# directories first if supplied, and search user-installed locations first
# so that we prefer user installs to system installs where both exist.
LIST(APPEND SUITESPARSE_CHECK_INCLUDE_DIRS
  ${SUITESPARSE_INCLUDE_DIR_HINTS}
  /opt/local/include
  /opt/local/include/ufsparse # Mac OS X
  /usr/local/homebrew/include # Mac OS X
  /usr/local/include
  /usr/local/include/suitesparse
  /usr/include/suitesparse # Ubuntu
  /usr/include)
LIST(APPEND SUITESPARSE_CHECK_LIBRARY_DIRS
  ${SUITESPARSE_LIBRARY_DIR_HINTS}
  /opt/local/lib
  /opt/local/lib/ufsparse # Mac OS X
  /usr/local/homebrew/lib # Mac OS X
  /usr/local/lib
  /usr/local/lib/suitesparse
  /usr/lib/suitesparse # Ubuntu
  /usr/lib)

# Given the number of components of SuiteSparse, and to ensure that the
# automatic failure message generated by FindPackageHandleStandardArgs()
# when not all required components are found is helpful, we maintain a list
# of all variables that must be defined for SuiteSparse to be considered found.
UNSET(SUITESPARSE_FOUND_REQUIRED_VARS)

# BLAS.
FIND_PACKAGE(BLAS QUIET)
IF (NOT BLAS_FOUND)
  SUITESPARSE_REPORT_NOT_FOUND(
    "Did not find BLAS library (required for SuiteSparse).")
ENDIF (NOT BLAS_FOUND)
LIST(APPEND SUITESPARSE_FOUND_REQUIRED_VARS BLAS_FOUND)

# LAPACK.
FIND_PACKAGE(LAPACK QUIET)
IF (NOT LAPACK_FOUND)
  SUITESPARSE_REPORT_NOT_FOUND(
[... 435 lines cut ...]
      SET(SUITESPARSE_IS_BROKEN_SHARED_LINKING_UBUNTU_SYSTEM_VERSION TRUE)
      MESSAGE(STATUS "Found system install of SuiteSparse "
        "${SUITESPARSE_VERSION} running on Ubuntu, which has a known bug "
        "preventing linking of shared libraries (static linking unaffected).")
    ENDIF (LSB_DISTRIBUTOR_ID MATCHES "Ubuntu" AND
      SUITESPARSE_LIBRARIES MATCHES "/usr/lib/libamd")
  ENDIF (LSB_RELEASE_EXECUTABLE)
ENDIF (CMAKE_SYSTEM_NAME MATCHES "Linux" AND
  SUITESPARSE_VERSION VERSION_EQUAL 3.4.0)

# Handle REQUIRED and QUIET arguments to FIND_PACKAGE
INCLUDE(FindPackageHandleStandardArgs)
IF (SUITESPARSE_FOUND)
  FIND_PACKAGE_HANDLE_STANDARD_ARGS(SuiteSparse
    REQUIRED_VARS ${SUITESPARSE_FOUND_REQUIRED_VARS}
    VERSION_VAR SUITESPARSE_VERSION
    FAIL_MESSAGE "Failed to find some/all required components of SuiteSparse.")
ELSE (SUITESPARSE_FOUND)
  # Do not pass VERSION_VAR to FindPackageHandleStandardArgs() if we failed to
  # find SuiteSparse to avoid a confusing autogenerated failure message
  # that states 'not found (missing: FOO) (found version: x.y.z)'.
  FIND_PACKAGE_HANDLE_STANDARD_ARGS(SuiteSparse
    REQUIRED_VARS ${SUITESPARSE_FOUND_REQUIRED_VARS}
    FAIL_MESSAGE "Failed to find some/all required components of SuiteSparse.")
ENDIF (SUITESPARSE_FOUND)`

zssjh, 2018-07-26:
replace FindSuiteSparse.cmake with the file below
https://ceres-solver.googlesource.com/ceres-solver/+/b0a8731fcdde31e6c37a54e8c1e1c00f853c0d5c/cmake/FindSuiteSparse.cmake

gaotao19890725, 2020-09-09:
error: invalid texture reference： home/gaotao/ElasticFusion/Core/src/Cuda/convenience.cuh:68 when ./ElasticFusion -l dyson_lab.klg with TX2, can you give me some suggestions

closed 2022-05-27 by mp3guy