mp3guy/ElasticFusion #162: Pangolin X11: Indirect GLX rendering context obtained

Linzert, 2018-10-21:
Hi,
  The installation is sucessful, but when I run "./ElasticFusion -l dyson_lab.klg", something wrong happened. The error is:
"""""""""""""""""""""""
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.X11 Error: GLXBadFBConfig
Pangolin X11: Indirect GLX rendering context obtained
Segmentation fault (core dumped)
""""""""""""""""""""""""
The pangolin examples run correctly. I donnot know why this could happend.

brendanowens, 2018-10-31:
I have this issue too

Daichou, 2018-10-31:
Pangolin can not pass through ssh.
I think you should provide your environement, such as OS,opengl version.

brendanowens, 2018-10-31:
I am using a clean install Ubuntu 16.04 on a remote machine. I have attempted accessing it via SSH in addition to VNC and x2go to use the GUI.

If Pangolin cannot pass through SSH, does this restrict ElasticFusion to only run on local machines?

Daichou, 2018-10-31:
Yes, I think you should use ElasticFusion in local machine and with GPU support.

closed 2022-05-27 by mp3guy