mp3guy/ElasticFusion #164: Realsense Dataset

SunshineDou, 2018-11-18:
Hello
 Recently, I am doing some research on this project, our lab has the Realsense D435, I make a klg format dataset with it, But when I run it in ElasticFusion, It failed. What's the problem? Does "ElasticFusion" support Realsense?

Thank you very much!

Daichou, 2018-11-18:
ElasticFusion cannot support librealsense2 sdk. If you want to use Intel Realsense D435 you can use librealsense2 openni wrapper [link](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2) or use my PR version #161.

SunshineDou, 2018-11-20:
Thank you very much for your generous help. I have a doubt that if I just want to run Elasticfusion with the klg format dataset(converted by png_to_klg), whose rgbd pictures comes from Realsense D435, Do I need to turn on realsense option in GUI ? Do I need to modify the GUI files just as #161 said? which steps do I need to do ?
Thanks again!

SunshineDou, 2018-11-20:
yeh, I just run like the above command "./Elasticfusion -l dataset1.klg" the dataset1.klg was converted by "png_to_klg", its rgb and depth pictures are all from realsense D435, but it failed ,the outcome is:
Segment error(Core dumped)

but when I run the dyson_lab.klg, it succeeded
Thanks for your help sincerely !

yyxr75, 2018-11-27:
Sir, how did you make through in running klg demo? I just run the command 
`./ElasticFusion -l dyson_lab.klg`
but error showed up:
`Error: invalid texture reference: xxxx/convenience.cuh: 68`
And I consulted to the answer which told me to
[add "61" to Line24](https://github.com/mp3guy/ElasticFusion/issues/55)
But 61 is already there when I check that file
Would you mind to give me help with it?
So much appreciation!

yyxr75, 2018-11-27:
I found out demo runs successful after I installed the graphiz before openni2 installing

ddg195, 2018-12-11:
> I found out demo runs successful after I installed the graphiz before openni2 installing
should we install graphiz? I do not understand ,I whatched every issue but neither one can solve the same problem ;I can run ElasticFusion successfully,but when I run which under Semantic folder always fail，and the sametime ，after i run Semanticfusion--elasticfusion，my elasticfusion also fail.
Any suggestion ? many many thanks.

yyxr75, 2018-12-14:
> I found out demo runs successful after I installed the graphiz before openni2 installing
> should we install graphiz? I do not understand ,I whatched every issue but neither one can solve the same [...]

hi man,
I wish I can help but that's really out of my reach.
You may consult in SemanticFusion issues

whubaichuan, 2019-06-24:
> yeh, I just run like the above command "./Elasticfusion -l dataset1.klg" the dataset1.klg was converted by "png_to_klg", its rgb and depth pictures are all from realsense D435, but it failed ,the [...]

Hi, are you success in real-time construction with realsenseSDK2 and ElasticFusion???

whubaichuan, 2019-06-24:
> Thank you very much for your generous help. I have a doubt that if I just want to run Elasticfusion with the klg format dataset(converted by png_to_klg), whose rgbd pictures comes from Realsense [...]

@SunshineDou  Hi. Are you success in real-time running with realsenseSDK2 and ElasticFusion???

SilentUniverse, 2019-09-18:
> I found out demo runs successful after I installed the graphiz before openni2 installing
> should we install graphiz? I do not understand ,I whatched every issue but neither one can solve the same [...]

I run with realsense 435 .  I got this error. i dont know  why.
i can run ./ElasticFusion -l dyson_lab.klg. it is OK.

./ElasticFusion

Creating live capture... failed!
DeviceOpen using default: no devices found
Creating live capture... start
Intel RealSense D435 827312072722
Auto Exposure enable
This camera does not support Auto White Balance option.
Auto White Balance disable
terminate called after throwing an instance of 'rs2::backend_error'
  what():  set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
Aborted (core dumped)

And  GPUTest don't  work.

GeForce GTX 1660 Ti
Searching for the best thread/block configuration for your GPU...
Error: no kernel image is available for execution on the device: /home/hys/ElasticFusion/Core/src/Cuda/convenience.cuh:68

What should i do ?
Thank you for your generous help!

psychorush, 2019-12-06:
> I run with realsense 435 . I got this error. i dont know why.
> i can run ./ElasticFusion -l dyson_lab.klg. it is OK.
>
> ./ElasticFusion
>
> Creating live capture... failed!
> DeviceOpen using default: no [...]

Hi!I meet the same problem.
Have you solved it?If so,how did you solve it?
Thank you for your help.

eduidl, 2019-12-15:
https://github.com/IntelRealSense/librealsense/issues/4586 may be related. In my environment, I downgraded linux kernel version from 5.0 to 4.18, and this problem was solved.

gaotao19890725, 2020-09-26:
Hi everyone:
I have installed the librealsense, and according to your suggestions in https://github.com/mp3guy/ElasticFusion/issues/164, I typed ‘ccmake ../’, then the following picture appears. But I do not know the meaning of “turning on Realsense option” (where is the realsense). Could you please tell me the way to turn on this and generate makefile, and how to run make. 
[image]

tureliuyaohua, 2020-10-29:
> yeh, I just run like the above command "./Elasticfusion -l dataset1.klg" the dataset1.klg was converted by "png_to_klg", its rgb and depth pictures are all from realsense D435, but it failed ,the [...]

是不是文件路径不对呢

wenjinshi, 2022-05-04:
I have the same problem with you. I can run the dataset Dyson_ Lab, but I try to convert the data produced by other cameras into KLG format, and the operation is like this: segment error, the core has been dumped

SilentUniverse, 2022-10-11:
谢谢