mp3guy/ElasticFusion #166: Running very slow in live

Ryan-Zirui-Wang, 2018-11-22:
Hi,

I am running on a Desktop with:
- i7-5960X 
- nvidia GeForce GTX 1080 Ti
- ubuntu 18.04 
- CUDA 9.2. 

It runs **perfectly** with the example dyson_lab.klg file but it runs very slow in live. I am using Orbbec Astra camera and I have double checked that all the dependencies have been build in Release. 

I noticed it output 
```
strs << "Your GPU \"" << dev << "\" isn't in the ICP Step performance database, please add it";
```
when running so I added the following code to `/ElasticFusion/Core/src/Utils/GPUConfig.h`
```
icpStepMap["GeForce GTX 1080 Ti"] = std::pair<int, int>(192, 112);
rgbStepMap["GeForce GTX 1080 Ti"] = std::pair<int, int>(112, 128);
rgbResMap["GeForce GTX 1080 Ti"] = std::pair<int, int>(224, 224);
so3StepMap["GeForce GTX 1080 Ti"] = std::pair<int, int>(128, 96);
```

which is the output when I run the GPUTest target but it is still running very slow in **live** (like 2-3fps).

Any ideas? Did I do anything obviously wrong?

Ryan-Zirui-Wang, 2019-01-21:
@xingxing4182 Nope, I don't use it anymore.

xingxing4182, 2019-01-30:
Thank you very much,do you know what the reason about it?

Ryan-Zirui-Wang, 2019-02-01:
@xingxing4182 , no, I would have posted my ideas here if I found out.

closed 2020-08-15 by ziruiw-dev