mp3guy/ElasticFusion #167: classroom datasets

SunshineDou, 2018-12-01:
Hello，
as I am doing a  project about 3D reconstruction, My teacher asked me to reconstruct the classroom, our lab has the realsense D435, and I have recorded the file and use the png_to_klg to convert it to klg format file, but the result is not  ideal, I cannot reconstruct the classroom. I don't know what's the problem. can you give me some help on the classroom dataset? or if there are available classroom dataset that can provide to me
Thank you very much~

scipione1, 2019-01-21:
I think I have the same problem. When running, the frames of my dataset are displayed really fast and no reconstruction appears. Have you solved the problem?

myc00, 2022-05-27:
> hello, I have the same problem, have you solved this problem？when i run the GUI, only frames of my dataset are displayed on the interface,  no pointcloud and the .ply is just 251 kb

myc00, 2022-05-27:
> I think I have the same problem. When running, the frames of my dataset are displayed really fast, even though I was really careful to move slowly when recording a dataset, and no reconstruction appears. Have you solved the problem?

hello, I have the same problem, have you solved this problem？when i run the GUI, only frames of my dataset are displayed on the interface, no pointcloud and the .ply is just 251 kb