mp3guy/ElasticFusion #169: ./ElasticFusion -l dyson_lab.klg error:

circleyuanquan, 2019-01-26:
Error: invalid texture reference: /home/turtlebot/ElasticFusion/Core/src/Cuda/convenience.cuh:68

sunit1409, 2019-02-13:
ElasticFusion/Core/src/CMakeLists.txt

add "61" to Line24:

```
set(CUDA_ARCH_BIN "20 30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch     to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) o    r 13 21(13)")
```

sunit1409, 2019-02-14:
try with sudo

sumitsarkar1, 2019-08-29:
my ElasticFusion/Core/src/CMakeLists.txt contains : set(CUDA_ARCH_BIN "30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) or 13 21(13)") and my card is RTX 2070 MAX Q with 418 driver on Ubuntu 16.04. Elasticfusion shows the same error i.e. "invalid texture reference" with dysollab kgl file.Please help.