mp3guy/ElasticFusion #182: Bug in rgbd Jacobian?

caselitz, 2019-05-29:
Hi @mp3guy,

I recently implemented some camera tracking based on the paper _Real-Time Visual Odometry from Dense RGB-D Images_ (Steinbruecker et al.) and looked into the ElasticFusion code which (I think) is also based on the Jacobian derived there. There is one point I don't quite understand and are wondering whether it might be a bug in the code. Im talking about this line:
https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/Cuda/reduce.cu#L535

`cloud` contains the original points of the last depth frame ([see here](https://github.com/mp3guy/ElasticFusion/blob/master/Core/src/Utils/RGBDOdometry.cpp#L409)), but I think it should be the warped points, i.e., transformed with the current estimate. I'm aware that the chosen points from the `cloud` change over the iterations due to data association (stored in `corresp`), but shouldn't these points also be transformed/warped into the next frame?

Here, I compared the paper (first part) and the ElasticFusion implementation (second part):
https://www.wolframcloud.com/obj/f63318f8-e5fb-4208-8a1e-9b36dc9ee3f5
You can see that they are identical if you set in the paper solution:
```
G1 = r11*X + r12*Y + r13*Z + tx
G2 = r21*X + r22*Y + r23*Z + ty
G3 = r31*X + r32*Y + r33*Z + tz
```
The code sets G is as a point from the unwarped `cloud`, but - as seen above - should use the warped point in my opinion.

I found another implementation, where I think this is also the case:
https://github.com/LucasSheng/dense-rgbd-visual-odometry/blob/master/calculateWarpingJacobian.m

In my own implementation I tried both versions and I don't really noticed much difference, I guess because the deltas between two poses are pretty small. If I understand correctly, the potential bug is basically "not to improve the linearization point with the current estimate" (but keeping the initial one). Maybe it was never noticed because it's no so significant, but I'd like to know what's correct from the theory...

Would appreciate if you could have a look and point me to my misunderstanding in case it's actually not a bug. 

Best, Tim

caselitz, 2019-05-29:
I looked at the paper again and think that I got it wrong, i.e., there is no bug. Eq. (16) and (21) of the paper say the linearization point is at `t0`, so the point must be warped with `g(t0)` and not - like I falsely assumed - with `g(t1)`. Since `g(t0)=id` this warp to be applied is actually a no-warp: as done in the code.

I now think the other implementation I referenced might have a bug... ;-)

closed 2019-05-29 by caselitz

heiwang1997, 2020-10-20:
Hi @caselitz , I don't think with the current implementation, linearization point is fixed to `g(t0)` during the iterations because if you look at [here](https://github.com/mp3guy/ElasticFusion/blob/8a60ca8e9d084be46e7130cbb7d7aa5d45b787af/Core/src/Utils/RGBDOdometry.cpp#L442), the correspondence between source and target images changes during each iteration, and the derivative computed will be thus different.
Another implmentation I found called InfiniTAM also used warpped points: [link](https://github.com/victorprad/InfiniTAM/blob/4156547e7f118c3a303d0b3c20f4da2aa3bfc9c9/InfiniTAM/ITMLib/Trackers/Shared/ITMColorTracker_Shared.h#L47).

Can you elaborate your answer a little bit more? I also find this part confusing...

caselitz, 2020-10-21:
Hi @heiwang1997,

right, one iteration of ICP performs data association and optimization: the data association defines the problem (error function) and the optimization solves it. Changing the correspondences (new data association in the next ICP iteration) makes a new error function and does not simply change the linearization point of the old one. So yes, the new/different function has another derivative.

Just consider each iteration independently regarding optimization, it's only that the previous iteration forms the new problem (via the new data association). But the linearization point is always (in every iteration) `g(t0)=id`.

That's my understanding, does this make sense to you?

Best,
Tim

heiwang1997, 2020-10-21:
Hi Tim,

Thank you so much for your answer. I now understand it that the derivatives are computed at `t0`. 
It seems that I mistakenly took the $\xi$ mentioned in the ICCV 2011 workshop paper as the camera pose, but it should actually be the inverse of camera pose. So in the implementation we actually warp the points at `t1` to the previous frame `t0`.

Best,
heiwang1997