mp3guy/ElasticFusion #187: ./ElasticFusion error

limyoona1996, 2019-07-14:
I use ElasticFusion wtih Ubuntu16.04 and Kinectv2. The Protonect in libfreenect2 and NiViewer in OpenNI2 can be run successfully.But when I ran the ElasticFusion. It turns out like this: 
sudo ./ElasticFusion
[sudo] password for ajon-1: 
Creating live capture... [Info] [Freenect2Impl] enumerating devices...
[Info] [Freenect2Impl] 11 usb devices connected
[Info] [Freenect2Impl] found valid Kinect v2 @4:3 with serial 000328144247
[Info] [Freenect2Impl] found 1 devices
libva info: VA-API version 0.39.0
libva info: va_getDriverName() returns -1
libva error: va_getDriverName() failed with unknown libva error,driver_name=(null)
[Error] [VaapiRgbPacketProcessorImpl] vaInitialize(display, &major_ver, &minor_ver): unknown libva error
[Info] [Freenect2DeviceImpl] opening...
[Info] [Freenect2DeviceImpl] transfer pool sizes rgb: 20*16384 ir: 60*8*33792
[Info] [Freenect2DeviceImpl] opened
success!
Waiting for first frame...................................................................................
  Could  anybody tell me what makes that error ? Thanks for the help!

limyoona1996, 2019-07-14:
The elasticfusion stops at "Waiting for first frame.." and does not continue.

limyoona1996, 2019-07-14:
Finally，I use these instructions to solve my problems:
    In libfreenect/build, 
 sudo apt-get install openni2-utils
 sudo make install-openni2
 NiViewer2
And then the ElasticFusion works,but I do not know why to do these instructions,could anybody tell me?

BloodLemonS, 2021-11-19:
> Finally，I use these instructions to solve my problems: In libfreenect/build, sudo apt-get install openni2-utils sudo make install-openni2 NiViewer2 And then the ElasticFusion works,but I do not know why to do these instructions,could anybody tell me?

What is your specific solution? I'm also waiting for the first frame

closed 2022-05-27 by mp3guy