mp3guy/ElasticFusion #188: error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES, and I have no Optimus GPU

JennyVVV, 2019-07-22:
Hello, everyone, when I run `./ElasticFusion -l dyson_lab.klg`, I got errors as follows:
Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/draw_global_surface.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/draw_global_surface.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/draw_global_surface_phong.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/quad.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/fxaa.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

OpenGL Error: XX (500)
In: /home/jenny/ElasticFusion/deps/Pangolin/include/pangolin/gl/gl.hpp, line 203
GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/quad.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/resize.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/quad.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/resize.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/quad.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/resize.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

OpenGL Error: XX (500)
In: /home/jenny/ElasticFusion/deps/Pangolin/include/pangolin/gl/gl.hpp, line 618
GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/index_map.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/index_map.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/empty.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/quad.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/visualise_textures.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

OpenGL Error: XX (500)
In: /home/jenny/ElasticFusion/deps/Pangolin/include/pangolin/gl/gl.hpp, line 203
GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/splat.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/depth_splat.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/splat.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/combo_splat.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/init_unstable.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/draw_feedback.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/draw_feedback.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/draw_global_surface.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/draw_global_surface.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/draw_global_surface.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/data.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/data.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/data.frag:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/update.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/copy_unstable.vert:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

GLSL Shader compilation failed: /home/jenny/ElasticFusion/Core/src/Shaders/copy_unstable.geom:
0:19(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, and 3.00 ES

OpenGL Error: XX (500)
In: /home/jenny/ElasticFusion/deps/Pangolin/include/pangolin/gl/gl.hpp, line 203
OpenGL Error: XX (501)
In: /home/jenny/ElasticFusion/deps/Pangolin/include/pangolin/gl/gl.hpp, line 203
Segmentation fault

Then I follow the installation issues, firstly, I check whether my computer supports Optimus, I run
` lspci -vnn | grep '\''[030[02]\]'`
and output 
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation Device [10de:1f02] (rev a1) (prog-if 00 [VGA controller])
which just one card. my os is ubantu16.04, NVIDIA Geforce RTX2070, cuda 9.0,  Driver Version: 430.34.  The Core, GPUTest, GUI have make successfully, what should I do next?

OldAAAA, 2019-11-02:
The same error as you, how can we solve that

YYYMagic, 2020-01-25:
I once had the same error.
I have two Graphics cards in my computer, NVIDIA and Intel. This error can be solved by using NVIDIA graphics card. You may set your graphics card in NVIDIA X Server Settings.

heiwuya-heiwuya, 2020-09-08:
Hello, How can you solve this problem? I just have one Graphiccard (GTX1660Ti 6GB).

YYYMagic, 2020-09-08:
> Hello, How can you solve this problem? I just have one Graphiccard (GTX1660Ti 6GB).

find NVIDIA X Server Settings, and then select use NVIDIA graphics card only.

heiwuya-heiwuya, 2020-09-09:
> > Hello, How can you solve this problem? I just have one Graphiccard (GTX1660Ti 6GB).
> 
> find NVIDIA X Server Settings, and then select use NVIDIA graphics card only.

Thank you for your answering. but i can`t find choose NVIDIA/Intel. I have no "PRIME Profiles" in Nvidia  x server settings

heiwuya-heiwuya, 2020-09-09:
> > Hello, How can you solve this problem? I just have one Graphiccard (GTX1660Ti 6GB).
> 
> find NVIDIA X Server Settings, and then select use NVIDIA graphics card only.

my Laptop always used Nvidia card(GTX1660Ti). it shows no intel card. when i ran elasticfusion, it shows also OpenGL Error: XX (1280)

xiaohang212, 2020-10-11:
MESA_GL_VERSION_OVERRIDE=3.3 ./ElasticFusion  ？？？？

BruceWANGDi, 2020-10-28:
@OldAAAA Have you figured out any solution to solve this issue? I met the same problem as you.

BruceWANGDi, 2020-10-28:
@JennyVVV Hi, I met the same issue, but my CUDA is 10.0. Could you tell me which solution you figured out to solve this problem?

BruceWANGDi, 2020-10-28:
@xiaohang212 Hi, I used the order you proposed, but there is still some issues. It said
 'Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.[Info] [DepthPacketStreamParser] 7 packets were lost' and
'OpenGL Error 501: GL_INVALID_VALUE: A numeric argument is out of range. In: /usr/local/include/pangolin/gl/gl.hpp, line 205'
Have you met this issue before?

heruiyuan, 2020-12-02:
I have the same issue but then I reinstall my Nvidia card driver and Cuda, it works!    https://github.com/mp3guy/ElasticFusion/issues/96 @JennyVVV

closed 2022-05-27 by mp3guy