mp3guy/ElasticFusion #189: error C2665: 'Uniform::Uniform' : none of the 7 overloads could convert all the argument types

kcshum, 2019-07-31:
I got errors when building Core. I tried the method provided(https://github.com/mp3guy/ElasticFusion/issues/63#issuecomment-261780789) by @antithing. But there are still errors after recompiling Pangolin and Suitesparse using vs2013 update 5.

error C2665: 'Uniform::Uniform' : none of the 7 overloads could convert all the argument types 
IntelliSense: no instance of constructor "Uniform::Uniform" matches the argument list argument types are: (const char [4], pangolin::OpenGlMatrix)

Did anybody fix this? Do I need to recompile other things?

Waiting for your reply.

kcshum, 2019-08-08:
Fixed.

closed 2019-08-08 by kcshum

skychenjian, 2020-03-15:
> 
> 
> Fixed.

Hello, I faced the same error: C2665. I tried to recompiling all of dependencies in vs2013, and rebuilding the Core. But still failed!
Could you share with me how you fixed this issue?