mp3guy/ElasticFusion #190: Elastic fusion issues RTX 2070 Max Q

sumitsarkar1, 2019-08-29:
my ElasticFusion/Core/src/CMakeLists.txt contains : set(CUDA_ARCH_BIN "30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) or 13 21(13)") and my card is RTX 2070 MAX Q with 418 driver on Ubuntu 16.04. Elasticfusion shows the same error i.e. "invalid texture reference" with dysonlab kgl file.Please help.

YabinXuTUD, 2019-10-30:
Hi, I have the same issue, have you solved it?

sumitsarkar1, 2019-10-30:
no i have not yet been able to solve it

YabinXuTUD, 2019-11-01:
Hi, I have solved this problem. Try this:
install cuda10.0 with your GPU;
then modify 61 to 75 in the codeline below in CMakeLists.txt
set(CUDA_ARCH_BIN "30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) or 13 21(13)")
Hope this can help you out!

sumitsarkar1, 2019-11-01:
thanx a ton..let me try it...hope you dont mind mailing you if i face any
problem

YabinXuTUD, 2019-11-03:
> thanx a ton..let me try it...hope you dont mind mailing you if i face any problem
> […](#)

Good luck and willing to help

sumitsarkar1, 2019-12-21:
The problem is not resolved yet.
1.) I installed cuda 10 on freshly installed ubuntu 16.04. But when I run
nvcc --version it says nvcc not installed. However synaptic package manager
shows nvcc 10 has been installed.
2.) The whole library builds with no error. The libfusion.so is build in
Core/build and executable ElasticFusion is build inside GUI/build
3.) When I try to run on dyslonlab example same texture error comes up
4.) I changed 61 to 75 and did cmake , make clean , make -j8 but no
luck...however when I run cmake ../src on the CMakeList.txt file inside
Core/src it shows me this :

Found SuiteSparse
-- CUDA NVCC target flags:
-gencode;arch=compute_30,code=sm_30;-gencode;arch=compute_35,code=sm_35;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_52,code=sm_52;-gencode;arch=compute_61,code=sm_61
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sumit/ElasticFusion/Core/build

Does this mean the change from 61 to 75 was not taken care by CMakeList.txt.

isunLt, 2020-05-14:
@sumitsarkar1 I have met the same problem as you. Now I have solved it. I'm use RTX2080MaxQ with CUDA10.0. I just move to ElasticFusion/Core/build and ccmake ../src/ . Change the  CUDA_ARCH_BIN and CUDA_ARCH_PTX to 75 and then rebuild. The problem is solved.

XunshanMan, 2020-06-12:
> @sumitsarkar1 I have met the same problem as you. Now I have solved it. I'm use RTX2080MaxQ with CUDA10.0. I just move to ElasticFusion/Core/build and ccmake ../src/ . Change the CUDA_ARCH_BIN and CUDA_ARCH_PTX to 75 and then rebuild. The problem is solved.

Hi @isunLt , I changed the CUDA_ARCH_BIN to 75, and then error occured:

> /home/xxx/workspace/ElasticFusion-master/Core/src/Cuda/reduce.cu(94): **error: identifier "__shfl_down" is undefined**
errors detected in the compilation of "/tmp/tmpxft_00005b0a_00000000-11_reduce.compute_75.cpp1.ii".
CMake Error at cuda_compile_generated_reduce.cu.o.cmake:266 (message):
Error generating file
/home/xxx/workspace/ElasticFusion-master/Core/build/CMakeFiles/cuda_compile.dir/Cuda/./cuda_compile_generated_reduce.cu.o
CMakeFiles/efusion.dir/build.make:70: recipe for target 'CMakeFiles/cuda_compile.dir/Cuda/cuda_compile_generated_reduce.cu.o' failed

I found that from CUDA 8.0+, the keyword **__shfl_down** has been depreciated. Since you are using CUDA10.0, I'm wondering why it worked on your PC.

Finally, this(https://github.com/mp3guy/ElasticFusion/issues/193) saved me, just annotate the line : 
> set(CUDA_ARCH_BIN "30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) or 13 21(13)")

I know it's weird but it worked for me....

By the way, the changes in CMakeLists.txt will only take effect when you use:
> rm -rf build && mkdir build
> cmake && make

sco09, 2021-02-18:
@XunshanMan I am using cuda 10.1 with 2070 maxq and met the same problem you described. And annotate that line works, thanks a lot

sco09, 2021-02-18:
I have done the test, if comment out that line, 
`set(CUDA_ARCH_BIN "30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) or 13 21(13)")`
cuda 11 could also work

closed 2022-05-27 by mp3guy