mp3guy/ElasticFusion #193: error: invalid texture reference:~/ElasticFusion/Core/src/cuda/convenience.cuh:68

SilentUniverse, 2019-09-18:
`sudo ./ElasticFusion -l dyson_lab.klg`
Error: invalid texture reference: /home/hys/ElasticFusion/Core/src/Cuda/convenience.cuh:68
I got this error. I dont know why?
how can i fix it?

sathiz91, 2019-09-19:
I also got the same Error. I'm using ubuntu16.04, cuda8 RTX2070

sathiz91, 2019-09-19:
Remove the following Line from /core/src/CMakeLists.txt 

**set(CUDA_ARCH_BIN "30 35 50 52 61" CACHE STRING "Specify 'real' GPU arch to build binaries for, BIN(PTX) format is supported. Example: 1.3 2.1(1.3) or 13 21(13)")** 

And build again. It solved the same issue.

gaotao19890725, 2020-09-09:
Dear sathiz91, I try your method, but it does not work, can you explain? Thank you.

avasalya, 2020-10-22:
@gaotao19890725  do the following in all the folders
1st Core, then GPUTest and then GUI

`rm -rf build && mkdir build && cd build && cmake ../src/ && make -j20
`

closed 2022-05-27 by mp3guy

SilentUniverse, 2022-05-27:
谢谢