mp3guy/ElasticFusion #197: What is the meaning of the matrix M when loading ground truth

qq1962572025, 2020-01-16:
When loading the ground truth ，the pose loaded is modified by M。
        Eigen::Matrix4f M;
      M <<  0,  0, 1, 0,
            -1,  0, 0, 0,
              0, -1, 0, 0,
             0,  0, 0, 1;

        pose = M.inverse() * camera_trajectory[timestamp] * M;
What does the Matrix M meas？

Viky397, 2021-02-16:
Hi! I'm also trying to pass in the groundtruth poses with the -p flag, but the camera frame upon running stays static and does not move according to what I'm passing in. Below is the format my pose.txt is in. Would you happen to know if I'm passing in incorrect data? Thanks!

```
1, -0.0875502211, 0.0817016153, 0.0000000000, 0.0000000000, 0.0000000000, 0.0200000000, 1.0000000000

2, -0.0875501915, 0.0817015510, 0.0000000000, 0.0000000000, 0.0000000000, 0.0200000000, 1.0000000000
```

nicolashei1201, 2021-07-17:
I figured it out 
the -p parameter allow user to input trajectory file, and M matrix will transfer the rotation into iSAM format
If your trajectory's format is set as the format that the code default, it shouldn't be iSAM format, so just disable the M matrix

closed 2022-05-27 by mp3guy