mp3guy/ElasticFusion #200: Error: context is destroyed: /home/bob/ElasticFusion/ElasticFusion/Core/src/Cuda/convenience.cuh:68

JannisEgelhof, 2020-02-21:
I use the Core API.
I could create a Pangolin Context.
I could call processFrame().

The error occurs here: 
ElasticFusion.cpp:processFrame(): `frameToModel.initFirstRGB(textures[GPUTexture::RGB]);`
RGBDOdometry.cpp:initFirstRGB(): `imageBGRToIntensity(textPtr, lastNextImage[0]);`
cudafuncs.cu:imageBGRToIntensity(): `cudaSafeCall(cudaBindTextureToArray(inTex, cuArr));`

Maybe it has to do something with the opengl context. #199 
Or some cuda configuration when building. #52 

Any suggestions?

JannisEgelhof, 2020-02-27:
Solved it. As mentioned in the manual you need to create an OpenGL Context before creating an ElasticFusion Object. 
If this runs in a different thread than your processFrame call you might get a problem. 
It is needed to "unbind" the context. In Pangolin it is done with

`pangolin::GetBoundWindow()->RemoveCurrent();`

Look also here: https://github.com/stevenlovegrove/Pangolin/blob/master/examples/HelloPangolinThreads/main.cpp

closed 2020-02-27 by JannisEgelhof

18721258653, 2021-11-08:
Thanks for your sharing very much~~
Could you please tell me how was your problem solved?I'm not sure what to change in the code.