mp3guy/ElasticFusion #205: Invalid surfels in ElasticFusion::downloadMap()

JannisEgelhof, 2020-04-02:
I make my own visualisation with the results of `downloadMap` using OpenGL.
I loop through the results of downloadmap and draw points with the positions of surfels with confidence > 10. This is the result I get (left side my visualization; right side the OpenGL texture from `        eFusion->getIndexMap().imageTex()->texture->RenderToViewport(true);`

[image]

I checked the surfels for confidence, however I always see these blue points everywhere.
Can there be an error in downloadMap? Is there a configuration which I should adapt when creating the ElasticFusion Object?

(At the moment the scene is mirrored and there are only points, no surfels in my visualization.)

JannisEgelhof, 2020-04-03:
Data from downloadMap was fine.
The circle was from normals being drawn. The blue area which you can see from mid to top-right is from colors interpreted as points.

closed 2020-04-03 by JannisEgelhof

Hyjale, 2020-06-26:
@a333klm Do you have any examples of code using just the Core API? Your other issues were very helpful as well.

a333klm, 2020-06-26:
`  pangolin::Params windowParams;
  windowParams.Set("SAMPLE_BUFFERS", 0);
  windowParams.Set("SAMPLES", 0);
  pangolin::CreateWindowAndBind("Main", 640, 480, pangolin::Params({ {
      "scheme", "headless"
    }
  }));
 ElasticFusion * eFusion = new ElasticFusion(timeDelta, countThresh, errThresh, covThresh,
                              closeLoops, iclnuim, reloc, photoThresh, confidence, depthCut,
                              icpThresh, fastOdom, fernThresh, so3, frameToFrameRGB, fileName);
while (newFramesAreAvailable) {
    unsigned char *color_ptr = ...
    unsigned char *depth_ptr = ...
    eFusion->processFrame(color_ptr, depth_ptr, timestamp);
    eFusion->predict();
    unsigned int numOfSurfels = eFusion->getGlobalModel().lastCount();
     Eigen::Vector4f *surfelmap =  eFusion->getGlobalModel().downloadMap();
`

Something like that. Those are the commands I use the most. Check the GUI implementation. There you can see how they use API to show the rendered stuff.

Hyjale, 2020-06-26:
```GLSL Shader compilation failed: /home/hyjale/ILLIXR_test/ILLIXR/ElasticFusion/Core/src/Shaders/empty.vert:

GLSL Shader compilation failed: /home/hyjale/ILLIXR_test/ILLIXR/ElasticFusion/Core/src/Shaders/quad.geom:

GLSL Shader compilation failed: /home/hyjale/ILLIXR_test/ILLIXR/ElasticFusion/Core/src/Shaders/resize.frag:

GLSL Program link failed: 
```
@a333klm Have you run into these errors? I am running the Core API in a separate thread and made sure to add in the code from your other posts.

a333klm, 2020-06-26:
Maybe I have run into such errors. I am not sure.
What do you mean with you run the Core API in a separate thread?

Did you create the Core in thread A and call processFrame in another thread?

Hyjale, 2020-06-26:
@a333klm Yes, I have both the Core and processFrame in separate threads.

a333klm, 2020-06-27:
Please consider that you need to reassign the OpenGL context then.

```
pangolin::Params windowParams;
windowParams.Set("SAMPLE_BUFFERS", 0);
windowParams.Set("SAMPLES", 0);
pangolin::CreateWindowAndBind("Main", 640, 480, pangolin::Params({ { "scheme", "headless" } }));
ElasticFusion * eFusion = new ElasticFusion(timeDelta, countThresh, errThresh, covThresh,
                              closeLoops, iclnuim, reloc, photoThresh, confidence, depthCut,
                              icpThresh, fastOdom, fernThresh, so3, frameToFrameRGB, fileName);

 pangolin::GetBoundWindow()->RemoveCurrent();
```
Check if context is available:
```
if (Pangolin::ShouldQuit)
{
    pangolin::BindToContext("Main");
}
while (newFramesAreAvailable) {
    unsigned char *color_ptr = ...
    unsigned char *depth_ptr = ...
    eFusion->processFrame(color_ptr, depth_ptr, timestamp);
    eFusion->predict();
    unsigned int numOfSurfels = eFusion->getGlobalModel().lastCount();
    Eigen::Vector4f *surfelmap =  eFusion->getGlobalModel().downloadMap();
```

Hyjale, 2020-06-30:
@a333klm My code structure looks similar to yours above and I'm also making sure that `pangolin::ShouldQuit()` is false before calling `processFrame()`. However, I'm still receiving a segfault when calling `processFrame`. Any thoughts?