mp3guy/ElasticFusion #206: How to create a surfel

zhaozhongch, 2020-04-04:
From the paper I know a surfel consists of position, normal, radius, timestamp...
However, I am curious how a surfel is created. Initially, we have an input image with 2D pixels and its correspondence depth. With these two we can get 3D points by project 2D pixel into the 3D world. Then I think a surfel should be constructed from those 3D points right?
My guess is that the position of a surfel is based on a bunch of 3D points that are near to each other, the paper may use the average position as the surfel's position.
But how to choose those points is a question. Those points are supposed to be on the same surface so that we can calculate its normal.
Correct me if my guess is wrong, if it is correct then how to choose points that may be on the same plane?
Thanks!

JannisEgelhof, 2020-04-08:
Hello,

> Then I think a surfel should be constructed from those 3D points right?
Yes, the surfels are constructed from those 3D points.

In their paper, they mention that their point fusion approach is based on Keller's approach: "Real-Time 3D Reconstruction in Dynamic Scenes Using Point-Based Fusion".
Look in the subchapter 4.1 Data Association.
They take an input pixel and check the points which are already in this area.

If they are too far away from the line between camera and pixel (viewing ray) they get discarded.
If the normals of these points do not fit to the normal of the input pixel they get discarded.
If there are more than 1 point left they pick the point which has been observed most often.
If multiple points have been observed they pick the one which is closest to the viewing ray.

> how to choose points that may be on the same plane?
I am not sure if the word "plane" makes sense here. They merge an existing point with a new observation.

> Also, how to initialize the radius and weight?
A surfel should be big enough to cover holes between surfels in the visualization. Look in "Dense Planar SLAM".
And the calculation of the weight is described in equation 10. This part is not obvious to me and I am not sure. I think the weight is not initialized like e.g. the radius. It is more like a placeholder.
You take one surfel x. This surfel has a corresponding node in the deformation graph. So does every surfel. Then a loop closure happens and your surfel x (among others) is deformed. You assign your 1,2,3 or 4 neighbors a weight according to their distance to x. Then you calculate the new position.

This is what I read in the paper and what I understood. Correct me if I am wrong. We can talk about it. It helps me too.

Viky397, 2021-01-19:
Hello! 
Thank you for the useful comments about the paper. 
Question: I have .klg files from my own dataset. However, when I run ElasticFusion -l <my.klg>, the GUI opens up and I can see my frames at the bottom, but no points appear and nothing seems to be happening. Any help is appreciated, thanks!

a333klm, 2021-01-19:
Maybe there is a playbutton hidden. I would compare your klg file with other klg files that work if that's possible.

Viky397, 2021-01-19:
Thank you for the feedback. I've compared it to the living room dataset from [ICL-NUIM](https://www.doc.ic.ac.uk/~ahanda/VaFRIC/iclnuim.html) and they are identically formatted, with the annotations.txt, rgb and depth folders. As well, when I play the ICL-NUIM klg (once I generate it), the depth is automatically visible in the GUI, whereas mine is not. Would you kindly be able to point me within the ElasticFusion code where it reads in the depth images (if at all possible). Perhaps I can do some digging. Thank you!

a333klm, 2021-01-20:
https://github.com/mp3guy/ElasticFusion/blob/8a60ca8e9d084be46e7130cbb7d7aa5d45b787af/GUI/src/Tools/LiveLogReader.cpp#L75

Viky397, 2021-01-20:
Thank you! I believe that it was because I didn't set my -depth larger, as my images are from a warehouse setting, so the default 3m is too small for them. i set -d to 100 and can visualize depth now. thanks!

Viky397, 2021-01-20:
However, my .freiburg output file only has pose values for the RGB image, not the depth. It looks something like this:

1.00000 0 0 0 4.3e-5 -0.00024 0.000456 1
It continues like this for all the rows.

Viky397, 2021-02-06:
Hello. My data is from a robot moving along the ground. However, when I run it through Elastic Fusion, the output trajectory veers off into the z-axis. This is impossible as the robot stays flat against the ground the entire time. I was wondering if there is any way for me to force ElasticFusion to only work in the XY plane and not optimize over Z. So keep z=0 the whole time. Thank you!

Viky397, 2021-02-16:
Hello. I have been trying to pass in ground truth poses, but when ElasticFusion runs, the camera frame stays static in the GUI, instead of following the trajectory I'm providing. This is what the first line of my pose.txt looks like:
` 1, -0.08755022111590, 0.08170161531600, 0.00000000000000, 0.00000000000000, 0.00000000000000, 0.01744086034340, 0.99984789662800`
Thank you

a333klm, 2021-02-16:
I have not used that function yet. 
I think I would check two things:

- Is there are button in the GUI where I can activate that the view should follow the camera?
- Given, I connect just the camera to ElasticFusion. Then I should be able to get the pose from the calculated position by ElasticFusion. There should be an API Function. Check what it outputs and replace the context of your pose.txt by that. Then run ElasticFusion again but this time with the input pose. Then you can verify if that function works. Or you can see whether the format of your pose.txt is correct.

That's what I would do. Maybe that does not make sense.

Viky397, 2021-02-16:
Hello, thank you for your reply!

There is no button in the GUI that activates following the camera.
As well, I was sure to check that the format of my pose.txt aligns with what is needed by ElasticFusion. 
`time/framenumber, x, y, z, qx, qy, qz, qw`

I even tried running it with the [ICL-NUIM](https://www.doc.ic.ac.uk/~ahanda/VaFRIC/iclnuim.html) dataset and their poses, but the GUI stays static as well, so it must not be my data at fault but something in the code...

closed 2022-05-27 by mp3guy