mp3guy/ElasticFusion #214: Segmentation fault (core dumped)

gaotao19890725, 2020-09-15:
At first, I successfully build the ElasticFusion, and when I run the command ./ElasticFusion -l dyson_lab.klg, it succeeds. But when I install the realsense2,  I run ./ElasticFusion or ./ElasticFusion -l dyson_lab.klg, the following error occurs:

Segmentation fault (core dumped)

Very strange, can you give me some concrete suggestion to modify it. Thanks a million.

heruiyuan, 2020-12-24:
Have you solved this problem? I have the same issue.

Viky397, 2021-01-15:
Same issue

closed 2022-05-27 by mp3guy

tfy14esa, 2022-07-04:
Hi @Viky397,
Were you able to resolve the issue? I am facing the same issue

mp3guy, 2022-07-04:
I would suggest building the binary in debug mode and then using gdb to get a stack trace, as a starting point.

tfy14esa, 2022-07-04:
> I would suggest building the binary in debug mode and then using gdb to get a stack trace, as a starting point.

Hi @mp3guy 
Thanks a lot for your response. See this thread for a complete stack trace: https://github.com/mp3guy/ElasticFusion/issues/222