mp3guy/ElasticFusion #220: compile err in tx2: unrecognized command line option "-msse2"

buenos-dan, 2020-12-13:
dji@manifold2:~/fusion/projects/ElasticFusion/Core/build$ cmake ../src/
-- Found BLAS: /usr/lib/libf77blas.so;/usr/lib/libatlas.so  
-- Found AMD library: /usr/lib/aarch64-linux-gnu/libamd.so
-- Found AMD header in: /usr/include/suitesparse
-- Found CAMD library: /usr/lib/aarch64-linux-gnu/libcamd.so
-- Found CAMD header in: /usr/include/suitesparse
-- Found COLAMD library: /usr/lib/aarch64-linux-gnu/libcolamd.so
-- Found COLAMD header in: /usr/include/suitesparse
-- Found CCOLAMD library: /usr/lib/aarch64-linux-gnu/libccolamd.so
-- Found CCOLAMD header in: /usr/include/suitesparse
-- Found CHOLMOD library: /usr/lib/aarch64-linux-gnu/libcholmod.so
-- Found CHOLMOD header in: /usr/include/suitesparse
-- Found SuiteSparseQR library: /usr/lib/aarch64-linux-gnu/libspqr.so
-- Found SuiteSparseQR header in: /usr/include/suitesparse
-- Found Intel Thread Building Blocks (TBB) library: /usr/lib/aarch64-linux-gnu/libtbb.so, assuming SuiteSparseQR was compiled with TBB.
-- Found Intel Thread Building Blocks (TBB) Malloc library: /usr/lib/aarch64-linux-gnu/libtbbmalloc.so
-- Found SuiteSparse_config library: /usr/lib/aarch64-linux-gnu/libsuitesparseconfig.so
-- Found SuiteSparse_config header in: /usr/include/suitesparse
-- Adding librt: /usr/lib/aarch64-linux-gnu/librt.so to SuiteSparse_config libraries (required on Linux & Unix [not OSX] if SuiteSparse is compiled with timing).
-- Did not find METIS library (optional SuiteSparse dependency)
-- Found SuiteSparse: /usr/include/suitesparse (found version "4.4.6") 
-- CUDA NVCC target flags: -gencode;arch=compute_30,code=sm_30;-gencode;arch=compute_35,code=sm_35;-gencode;arch=compute_50,code=sm_50;-gencode;arch=compute_52,code=sm_52;-gencode;arch=compute_61,code=sm_61
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dji/fusion/projects/ElasticFusion/Core/build
dji@manifold2:~/fusion/projects/ElasticFusion/Core/build$ make -j8
[  4%] Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/Cuda/cuda_compile_1_generated_reduce.cu.o
[  9%] Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/Cuda/cuda_compile_1_generated_cudafuncs.cu.o
In file included from /usr/local/cuda-9.0/include/host_config.h:50:0,
                 from /usr/local/cuda-9.0/include/cuda_runtime.h:78,
                 from <command-line>:0:
/usr/local/cuda-9.0/include/crt/host_config.h:119:2: error: #error -- unsupported GNU version! gcc versions later than 6 are not supported!
 #error -- unsupported GNU version! gcc versions later than 6 are not supported!
  ^~~~~
In file included from /usr/local/cuda-9.0/include/host_config.h:50:0,
                 from /usr/local/cuda-9.0/include/cuda_runtime.h:78,
                 from <command-line>:0:
/usr/local/cuda-9.0/include/crt/host_config.h:119:2: error: #error -- unsupported GNU version! gcc versions later than 6 are not supported!
 #error -- unsupported GNU version! gcc versions later than 6 are not supported!
  ^~~~~
CMake Error at cuda_compile_1_generated_reduce.cu.o.cmake:219 (message):
  Error generating
  /home/dji/fusion/projects/ElasticFusion/Core/build/CMakeFiles/cuda_compile_1.dir/Cuda/./cuda_compile_1_generated_reduce.cu.o

CMakeFiles/efusion.dir/build.make:70: recipe for target 'CMakeFiles/cuda_compile_1.dir/Cuda/cuda_compile_1_generated_reduce.cu.o' failed
make[2]: *** [CMakeFiles/cuda_compile_1.dir/Cuda/cuda_compile_1_generated_reduce.cu.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMake Error at cuda_compile_1_generated_cudafuncs.cu.o.cmake:219 (message):
  Error generating
  /home/dji/fusion/projects/ElasticFusion/Core/build/CMakeFiles/cuda_compile_1.dir/Cuda/./cuda_compile_1_generated_cudafuncs.cu.o

CMakeFiles/efusion.dir/build.make:63: recipe for target 'CMakeFiles/cuda_compile_1.dir/Cuda/cuda_compile_1_generated_cudafuncs.cu.o' failed
make[2]: *** [CMakeFiles/cuda_compile_1.dir/Cuda/cuda_compile_1_generated_cudafuncs.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/efusion.dir/all' failed
make[1]: *** [CMakeFiles/efusion.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
dji@manifold2:~/fusion/projects/ElasticFusion/Core/build$ 
dji@manifold2:~/fusion/projects/ElasticFusion/Core/build$ 
dji@manifold2:~/fusion/projects/ElasticFusion/Core/build$ make -j8
[  4%] Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/Cuda/cuda_compile_1_generated_cudafuncs.cu.o
[  9%] Building NVCC (Device) object CMakeFiles/cuda_compile_1.dir/Cuda/cuda_compile_1_generated_reduce.cu.o
/home/dji/fusion/projects/ElasticFusion/Core/src/Cuda/reduce.cu(94): warning: function "__shfl_down(float, unsigned int, int)"
/usr/local/cuda-9.0/include/sm_30_intrinsics.hpp(278): here was declared deprecated ("__shfl_down() is deprecated in favor of __shfl_down_sync() and may be removed in a future release (Use -Wno-deprecated-declarations to suppress this warning).")

/home/dji/fusion/projects/ElasticFusion/Core/src/Cuda/reduce.cu(95): warning: function "__shfl_down(float, unsigned int, int)"
/usr/local/cuda-9.0/include/sm_30_intrinsics.hpp(278): here was declared deprecated ("__shfl_down() is deprecated in favor of __shfl_down_sync() and may be removed in a future release (Use -Wno-deprecated-declarations to suppress this warning).")

/home/dji/fusion/projects/ElasticFusion/Core/src/Cuda/reduce.cu(96): warning: function "__shfl_down(float, unsigned int, int)"
/usr/local/cuda-9.0/include/sm_30_intrinsics.hpp(278): here was declared deprecated ("__shfl_down() is deprecated in favor of __shfl_down_sync() and may be removed in a future release (Use -Wno-deprecated-declarations to suppress this warning).")
[... 643 lines cut ...]
c++: error: unrecognized command line option ‘-msse2’
c++: error: unrecognized command line option ‘-msse3’
CMakeFiles/efusion.dir/build.make:128: recipe for target 'CMakeFiles/efusion.dir/GlobalModel.cpp.o' failed
make[2]: *** [CMakeFiles/efusion.dir/GlobalModel.cpp.o] Error 1
CMakeFiles/efusion.dir/build.make:141: recipe for target 'CMakeFiles/efusion.dir/IndexMap.cpp.o' failed
make[2]: *** [CMakeFiles/efusion.dir/IndexMap.cpp.o] Error 1
[ 40%] Building CXX object CMakeFiles/efusion.dir/Utils/CholeskyDecomp.cpp.o
[ 40%] Building CXX object CMakeFiles/efusion.dir/Utils/DeformationGraph.cpp.o
[ 45%] Building CXX object CMakeFiles/efusion.dir/Ferns.cpp.o
c++: error: unrecognized command line option ‘-msse2’
c++: error: unrecognized command line option ‘-msse3’
c++: error: unrecognized command line option ‘-msse2’
c++: error: unrecognized command line option ‘-msse3’
CMakeFiles/efusion.dir/build.make:102: recipe for target 'CMakeFiles/efusion.dir/Ferns.cpp.o' failed
make[2]: *** [CMakeFiles/efusion.dir/Ferns.cpp.o] Error 1
CMakeFiles/efusion.dir/build.make:154: recipe for target 'CMakeFiles/efusion.dir/Utils/CholeskyDecomp.cpp.o' failed
make[2]: *** [CMakeFiles/efusion.dir/Utils/CholeskyDecomp.cpp.o] Error 1
c++: error: unrecognized command line option ‘-msse2’
c++: error: unrecognized command line option ‘-msse3’
CMakeFiles/efusion.dir/build.make:167: recipe for target 'CMakeFiles/efusion.dir/Utils/DeformationGraph.cpp.o' failed
make[2]: *** [CMakeFiles/efusion.dir/Utils/DeformationGraph.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/efusion.dir/all' failed
make[1]: *** [CMakeFiles/efusion.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

tkircher, 2020-12-13:
@buenos-dan You need to use a version of gcc that isn't obsolete.

buenos-dan, 2020-12-13:
@tkircher gcc version is gcc-5, and the cpu is tx2(aarch64). I just known that -msse2 -msse3 is x86 compile option. Therefore, I delete these in CMakeLists.txt, and then I build the project succeed. Thanks for your comment.

closed 2022-05-27 by mp3guy