mp3guy/ElasticFusion #221: Short range depth

Docea, 2020-12-19:
Hello,

I'm trying to configure ElasticFusion to run with a valid depth range of between around 1cm and 75cm. It's clear how to reduce the maximum depth, but reducing the minimum depth has not beet so clear. Only more distant points seem to be usable, and the minimum depth threshold (which I can't find specified) prevents closer depths (<~50cm) from being used. My question is whether there is any way to reduce this minimum distance threshold?

Many thanks for any help!

Docea, 2021-01-23:
Depth is represented in uint, where the value represents the depth in millimeters. For .klg log files, the depth is converted to millimeter scale uint by the scripts which generate the .klg files. To get depth in a different scale, the scripts can be edited.

closed 2021-01-23 by Docea

Viky397, 2021-01-24:
Hello, I am trying to run my depth images, which are from a larger warehouse setting. When I run ElasticFusion, in the GUI at the bottom I see my rgb image, but the respective depth image looks quite distorted. As well, ElasticFusion cannot find any points and nothing really seems to be happening. 

Would you be able to kindly within the code point me to where it works on the input depth images? I would like to see if perhaps I need to do some normalization or scaling to my depth images before passing them in. Thank you!

Docea, 2021-01-24:
Hi, if you attach an image I might be able to help you a bit better, but if you're generating .klg files, check the script that does this because it encodes the input depth image into the .klg file.

Otherwise:
- /Core/src/Shaders/depth_bilateral.frag
- /Core/src/Shaders/depth_metric.frag
- /Core/src/Shaders/depth_norm.frag
- void ElasticFusion::normaliseDepth(const float & minVal, const float & maxVal) in /Core/src/ElasticFusion.cpp

Hope this helps!

Viky397, 2021-01-24:
Hi, thanks for the quick reply! Attached is an example of my depth image. I've been comparing it to the ICL-NUIM depth data, but can't find any discrepancies. The script I'm using to generate the klg's is from here, written for ElasticFusion: https://github.com/HTLife/png_to_klg 
[image]
.

Docea, 2021-01-24:
Is this a depth image that you create of the larger warehouse? Is this was it looks like before or after you pass it to ElasticFusion?

Viky397, 2021-01-24:
Correct, it was taken with an Intel Realsense RGBD camera. This is what I am passing into ElasticFusion. For some reason, i need to specify -d 100 in order for the depth image to even show up in the GUI when I run ElasticFusion, and when i do, the image looks like this at the bottom in the GUI:
[image]

Viky397, 2021-01-24:
When I compare what the ICL-NUIM depth image looks like raw, versus in the ElasticFusion GUI, they look the same.

Docea, 2021-01-25:
Have you increased the depth threshold for ElasticFusion? how big is this warehouse you're talking about?

To me it looks like the depth image that you created is formed badly. Depth images (as far as I know) can be saved with 32 bit or 16 bit representations. If you take 32bit data and save it as 16bit, you could get errors of the kind you're seeing.

Viky397, 2021-01-25:
When I run it in the command, I specify: `-d : Cutoff distance for depth processing (default 3m).` and I set it to be `-d 100`
The warehouse is large, at least 40 m, however, nothing shows in the GUI when I specify anything less than 90, which is a bit too large in my opinion.

Hm, the original images are stored in .bag files, and I use ROS to extract them, and they seem to be extracted as 16-bit

`rosrun image_view image_saver _encoding:=16UC1 image:=/left/realsense/aligned_depth_to_color/image_raw/ _filename_format:="%04i.png"`

Docea, 2021-01-25:
I'd imagine that the depth images you're using to construct the .klg files are the problem here. Does the depth image you generate match what you see in the corresponding RGB images?

Viky397, 2021-01-25:
Yep, they look like they align. 

[image]

Viky397, 2021-01-25:
Hm, actually, now that you mention it, it seems that the two sets of images don't align fully. I was looking closer to the start of the data when the camera wasn't moving much, but in the middle as it moves through the scene, the images are off by about 15 frames from eachother. 

Do you believe this may be why no points are being calculated by ElasticFusion and the main GUI in the middle of the screen remains blank when I run it?

Doesn't really solve the question of why the depth image in the GUI looks so different from the original though.

Docea, 2021-01-25:
Can you try to play the rosbag and view the disparity image with Image_viewer? Then compare what you see through image viewer to the depth images that you save. If they are different, your problem is here most likely.

rosbag play <path_to_bag>
rosrun image_view disparity_view image:=<disparity_topic>

Viky397, 2021-01-25:
hm when I run that I get the error:

Docea, 2021-01-25:
Ah, I am not so sure how to solve this because I've never worked with intel realsense, but I think it's worth persevering to find out. Or, you could try to find out what the datatype is for  "/left/realsense/aligned_depth_to_color/image_raw/ ", and compare it to what your are saving.

Viky397, 2021-01-25:
So when i `rostopic echo /left/realsense/aligned_depth_to_color/image_raw/`, values between 0-255 appear. But when the depth images are extracted and saved with `image_saver`, they are saved as uint16.

Docea, 2021-01-25:
In that case, are you sure that /left/realsense/aligned_depth_to_color/image_raw/ is a depth image? It also could be likely that rostopic echo simply prints the uint value of each byte in the image.

Viky397, 2021-01-25:
ah yes sorry, reading this thread it seems that echo converts the data to char

https://github.com/IntelRealSense/realsense-ros/issues/714

Viky397, 2021-01-25:
Sorry, so just for clarity, it seems my depth image is uint16, yet still not appearing correctly in the GUI. As well, ElasticFusion still is not picking up on any points. Would you happen to have any other tips? thank you!

Docea, 2021-01-25:
If I recall correctly, ElasticFusion only works with 16bit depth. It is taken in as 16bit uint, where the value represents the distance in whole millimeters. With this, the maximum depth encoded is 65536mm (65.536m). So, values for depth larger that are not possible unless you change the scale of the input. In my case I changed the scale through the .klg generating file such that the uint value represented the distance in 0.1mm rather than 1mm, giving me a maximum depth of 6.5536m. You might want to go the other way where the uint value represents the distance in 2mm increments, giving you up to ~130m. 

But, I still think your depth image looks a bit odd. On top of that, I think the intel realsense system wouldn't be able to capture accurate depth estimates much more than 20m - but I'm not sure.

Viky397, 2021-01-25:
It finally worked! It seems the big issue was probably the fact that the rgb and depth images were offset from eachother, the way they were being saved from the rosbag. 

One last question, is there a way to visualize the .freiburg or .ply files that ElasticFusion outputs?

Thank you so much again for all your help!
[image]

Docea, 2021-01-25:
Great to hear it started working :)

For viewing the trajectory (.freiburg), I just plot a line in 3D using matplotlib

For viewing the .ply, I am not sure. If you can convert to .pcd, you can view it with:
pcl_viewer -multiview 1 <pcd_filepath>

Viky397, 2021-01-25:
Hm, this is what my .ply file looks like, and I get the error `Read PLY failed: unable to open file: ` when I use open3d's `o3d.io.read_point_cloud`. Any thoughts? If not, that's totally fine, you've helped already so much :)
[image]

Docea, 2021-01-25:
Actually, I don't have much of an idea about how to process .ply files so I think we are at the limit of what I can help with here 😄 Maybe you could upload it to an online viewer?

Viky397, 2021-01-25:
I downloaded Meshlab and can view it on there. Thank you again so much :)

Docea, 2021-01-25:
No worries

Viky397, 2021-01-28:
Hi! Another quick question: Would it be at all possible to force the optimization to only happen in the XY plane, and force Z=0? Since my data is for sure in the XY plane only. Thanks!

Docea, 2021-01-31:
Hi, 

Sorry for the delayed reply, I had a deadline. Which optimisation? I'm not sure exactly what you mean?

Viky397, 2021-01-31:
No worries! My data is from a robot moving along the ground. However, when I run it through Elastic Fusion, the output trajectory veers off into the z-axis. Since this is impossible, I was wondering if there is any way for me to force ElasticFusion to only work in the XY plane.

Viky397, 2021-02-03:
Similarly, I've run another dataset through ElasticFusion that is sort of an obstacle course, and the resulting mesh and trajectory are in all directions, when they should only be in one plane, as the robot stays flat against the ground the entire time.

Docea, 2021-02-03:
I see what you mean now. It's a good idea, and I guess there must be some way to make ElasticFusion do that, but I don't know off the top of my head. If anything comes to mind, I'll let you know :)

Viky397, 2021-02-03:
Alrighty, thank you! :)

Viky397, 2021-02-06:
Sorry for the barrage of questions (you seem very well-versed in this paper), but would you happen to know where the geometric and photometric pose estimation is happening in the code? I've found where the deformation cost functions are, but can't seem to find either of the pose estimation or joint optimization cost functions. Thank you!

Docea, 2021-02-06:
No worries. That part of the code is in RGBDOdometry.(h & cpp), but it also
uses cuda functions. You'll be able to find them if you search the whole
project for the functions you're looking for (ctrl+shift+f in Atom, for me).

Viky397, 2021-02-06:
Thanks! These are the two functions I'm searching for, however they don't seem to be named accordingly. In DeformationGraph.cpp I'm able to see the rotation, regularization and constraint cost functions, since they are named as such, but perhaps I am looking for the wrong function name. Would you happen to recall what they are called? 
[image]
[image]

Docea, 2021-02-06:
Yup, so inside RGBDOdometry.cpp, inside getIncrementalTransform(...) you can find:
- computeRgbResidual(...) at line 442
- icpStep(...) at line 494
- rgbStep(...) at line 524

Each of these have to do with computing the error terms there.

I'm pretty sure the errors values themselves are rgbError, lastRGBError, lastICPError. All of those are inside getIncrementalTransform(...).

Inside Core/src/Cuda/reduce.cu:
- icpStep(...) is defined at line 418
- rgbStep(...) is defined at line 626

Those should each implement the functions computing the errors you're looking for. It's all done on GPU and I'm not so confident there. I think if you look at the 'search' and 'getProducts' functions inside of  'ICPReduction', and 'getProducts' inside of RGBReduction (all inside of reduce.cu), I think you might have some luck. Let me know how it goes!

Viky397, 2021-02-07:
Thank you for the in-depth response! It helps a lot :)

Viky397, 2021-02-10:
Hiya! I'm also trying to pass in the groundtruth poses with the -p flag, but the camera frame upon running stays static and does not move according to what I'm passing in. The z, qx and qy columns are always zero, and the qz column is always 0.02 as seen below. I'm thinking it might have something to do with the transform but I'd love to hear your opinion :)

`1, -0.0875502211, 0.0817016153, 0.0000000000, 0.0000000000, 0.0000000000, 0.0200000000, 1.0000000000
2, -0.0875501915, 0.0817015510, 0.0000000000, 0.0000000000, 0.0000000000, 0.0200000000, 1.0000000000`

Docea, 2021-02-11:
Hi,

When you run ElasticFusion on a dataset you should get an output trajectory file called <filename>.freiburg . I've not used the ground truth poses myself, but I guess the poses that you supply to ElasticFusion should have the same format as in that trajectory file.

Hope that helps!

Viky397, 2021-02-11:
Thanks! And yes, my poses do have that format (except EF also requires that they have commas between the columns), but for some reason the frame when I run ElasticFusion on these groundtruth poses that I'm providing, the frame in the GUI stays static and builds all the surfels in the same spot, when it should in fact be moving in the xy plane according to the pose file I'm providing. Quite odd...

wuqs-net, 2021-08-04:
Hello @Docea , thank you for pointing out the scaling of the depth map. 

> In my case I changed the scale through the .klg generating file such that the uint value represented the distance in 0.1mm rather than 1mm, giving me a maximum depth of 6.5536m.

I'm trying to generate a .klg with scale similar to yours (maximum depth around 5m). How do you change the scale in the code? What needs to be changed?