mp3guy/ElasticFusion #222: segmentation fault

wangmiaowei, 2020-12-29:


wangmiaowei, 2020-12-29:
Everything is OK but when I run the ./ElasticFusion -l dyson_lab.klg -g. withe several seconds, "segmentation fault (core dumped) occures"

tfy14esa, 2022-07-04:
> Everything is OK but when I run the ./ElasticFusion -l dyson_lab.klg -g. withe several seconds, "segmentation fault (core dumped) occures"

Hi @wangmiaowei 
Were you able to resolve the issue? I am facing the same error.

thachdo, 2022-10-25:
Hi @tfy14esa, I meet the same issue here. Have you resolved that?

mp3guy, 2022-10-25:
I can repro, investigating.

mp3guy, 2022-10-25:
I presume you're running on a laptop with NVIDIA Prime? You need to make sure `glxinfo` returns NVIDIA and not SGI (Mesa). I did: `sudo prime-select nvidia`

closed 2022-10-25 by mp3guy

thachdo, 2022-10-26:
Hi Thomas, it's running on a workstation, NVIDIA Quadro RTX 4000.  This is the output of `glxinfo`
```
server glx vendor string: NVIDIA Corporation
client glx vendor string: Mesa Project and SGI
    Vendor: Mesa/X.org (0xffffffff)
OpenGL vendor string: Mesa/X.org
``` 
Also, below is the backtrace
```
Thread 1 "ElasticFusion" received signal SIGSEGV, Segmentation fault.
_dl_close (_map=0x0) at ./elf/dl-close.c:795
795	./elf/dl-close.c: No such file or directory.
(gdb) backtrace
#0  _dl_close (_map=0x0) at ./elf/dl-close.c:795
#1  0x00007ffff75bac28 in __GI__dl_catch_exception (exception=exception@entry=0x7fffffffd3d0, 
    operate=<optimized out>, args=<optimized out>) at ./elf/dl-error-skeleton.c:208
#2  0x00007ffff75bacf3 in __GI__dl_catch_error (objname=0x7fffffffd428, errstring=0x7fffffffd430, 
    mallocedp=0x7fffffffd427, operate=<optimized out>, args=<optimized out>) at ./elf/dl-error-skeleton.c:227
#3  0x00007ffff74d61ae in _dlerror_run (operate=<optimized out>, args=<optimized out>) at ./dlfcn/dlerror.c:138
#4  0x00007ffff74d5ed8 in __dlclose (handle=<optimized out>) at ./dlfcn/dlclose.c:31
#5  0x00007fff5490ef81 in ?? () from /usr/lib/x86_64-linux-gnu/libcuda.so.1
#6  0x00007fff5490eff3 in ?? () from /usr/lib/x86_64-linux-gnu/libcuda.so.1
#7  0x00007fff549f9e02 in ?? () from /usr/lib/x86_64-linux-gnu/libcuda.so.1
#8  0x00007ffff7eeef98 in __cudart1359 () from /root/ElasticFusion/build/Core/libefusion.so
#9  0x00007ffff7f48b71 in cudaGraphicsGLRegisterImage () from /root/ElasticFusion/build/Core/libefusion.so
#10 0x00005555555883d8 in GUI::GUI(bool, bool) ()
#11 0x00005555555722fd in MainController::MainController(int, char**) ()
#12 0x000055555556f761 in main ()

```

mp3guy, 2022-10-26:
Well something is wrong with your drivers or configuration. You should be running X11. And it should not list Mesa or SGI anywhere in the glxinfo client or server. This is not a problem with ElasticFusion, it's a problem with your setup.

Ru-hulu, 2023-01-30:
Hi， I just have the same error
https://github.com/mp3guy/ElasticFusion/issues/222#issuecomment-1173679808. 
core dumped happened in the same line.
I tried to use this command "glxinfo", and output was like:
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
.......
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
client glx extensions:
.......
https://github.com/mp3guy/ElasticFusion/issues/222#issuecomment-1291831038 said, it should not list Mesa or SGI, but I don't know how to make it work.
By the way, nvidia-smi:
NVIDIA-SMI 520.61.05    Driver Version: 520.61.05    CUDA Version: 11.8
in Ubuntu 22.04

I am not sure how to solve this problem, could you give me some suggestions?
https://github.com/mp3guy/ElasticFusion/issues/222#issuecomment-1291831038

mp3guy, 2023-01-30:
Are you running on a Laptop? 

Either set the default GPU via the Prime settings in `nvidia-settings`, or run `sudo prime-select nvidia` and reboot.

Ru-hulu, 2023-01-30:
https://github.com/mp3guy/ElasticFusion/issues/222#issuecomment-1408493404
I tried and I worked.
Perhaps I forgot to reboot my laptop.
Thank you！！！-v-