mp3guy/ElasticFusion #228: Fit it on realsense l515 device

Sylvia6, 2021-04-08:
I'm adapting Elasticfusion  to the l515 device of realsense. At present, the device can be detected, but there are some problems with my Elasticfusion result.

Considering that the RGB resolution of the l515 device is a fixed combination of 1280 * 720 and the depth resolution of 640 * 480, I want to know how to make elasticfusion work correctly. Can u help me?

Sylvia6, 2021-04-08:
Here is my elasticfusion result:
[image]

As the picture shows, the rgb and depth result is dismatched with the core interface

closed 2021-04-13 by Sylvia6

YangXJ95, 2022-05-12:
Hello @Sylvia6, did you solve this mismatch problem in the end?