mp3guy/ElasticFusion #233: Error when building the ElasticFusion GUI

MichaelMano1, 2022-02-24:
Hi,when I build the package for GUI,I encountered with this problem,does anyone have the same issue?

/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp: In constructor ‘MainController::MainController(int, char**)’:
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:115:28: error: ‘class std::shared_ptr<pangolin::VarValueT<bool> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->flipColors->Ref().Set(logReader->flipColors);
                                                  ^~~
                                                  get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:116:25: error: ‘class std::shared_ptr<pangolin::VarValueT<bool> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->rgbOnly->Ref().Set(false);
                                                 ^~~
                                                 get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:117:25: error: ‘class std::shared_ptr<pangolin::VarValueT<bool> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->pyramid->Ref().Set(true);
                                                 ^~~
                                                 get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:118:26: error: ‘class std::shared_ptr<pangolin::VarValueT<bool> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->fastOdom->Ref().Set(fastOdom);
                                                  ^~~
                                                  get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:119:37: error: ‘class std::shared_ptr<pangolin::VarValueT<float> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->confidenceThreshold->Ref().Set(confidence);
                                                               ^~~
                                                               get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:120:29: error: ‘class std::shared_ptr<pangolin::VarValueT<float> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->depthCutoff->Ref().Set(depth);
                                             ^~~
                                             get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:121:27: error: ‘class std::shared_ptr<pangolin::VarValueT<float> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->icpWeight->Ref().Set(icp);
                                                   ^~~
                                                   get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:122:21: error: ‘class std::shared_ptr<pangolin::VarValueT<bool> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->so3->Ref().Set(so3);
                                     ^~~
                                     get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:123:33: error: ‘class std::shared_ptr<pangolin::VarValueT<bool> >’ has no member named ‘Set’; did you mean ‘get’?
     gui->frameToFrameRGB->Ref().Set(frameToFrameRGB);
                                                           ^~~
                                                           get
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp: In member function ‘void MainController::run()’:

After I change the "Set" to "get",still have problem:

/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp: In member function ‘void MainController::run()’:
/home/cnty/elastic_bridge_dic/src/elastic_bridge/deps/ElasticFusion/GUI/src/MainController.cpp:335:48: error: no matching function for call to ‘std::shared_ptr<pangolin::VarValueT<std::__cxx11::basic_string<char> > >::get(std::__cxx11::basic_stringstream<char>::__string_type)’
         gui->trackInliers->Ref().get(stri.str());
                                                                                  ^

wenjinshi, 2022-02-25:
I also encountered this problem. Has your problem been solved？

I try to change the symbol " ."   before  Set   to  "->",then make it ,but when I use the dyson_lab.klg,there are new errors, so I am not sure is this change right?

new error:

erminate called after throwing an instance of 'pangolin::FactoryRegistry::ParameterMismatchException'
  what():  Unable to open URI
  full_uri: 
  scheme: default
  params:
    SAMPLE_BUFFERS = 0
    SAMPLES = 0
    w = 1485
    h = 980
    window_title = Main
 Unrecognized options for scheme (SAMPLES,SAMPLE_BUFFERS,)

MichaelMano1, 2022-02-26:
@wenjinshi  May be its the problem with Pangolin version.The old version like 0.5 was build with c++11,the lastest version like 0.6 ,is build with c++17.After I change the Pangolin version to 0.5.The GUI problem is solved , but this time I can't build Core package.It seems Core package require Pangolin 0.6,and GUI require Pangolin 0.5.

wenjinshi, 2022-02-26:
Thank  you

maybe you can try my  mthod to avoid the error, but I am not sure weather  does it cause other errors.

As for your advice， I have noticed there are some people who have successfully run the dataset using the same pangolin.

could you tell me your version of Ubuntu?

wenjinshi, 2022-02-27:
could you tell me the last step? How can I solve my issue,thank you a lot

terminate called after throwing an instance of 'pangolin::FactoryRegistry::ParameterMismatchException'
  what():  Unable to open URI
  full_uri: 
  scheme: default
  params:
    SAMPLE_BUFFERS = 0
    SAMPLES = 0
    w = 1485
    h = 980
    window_title = Main
 Unrecognized options for scheme (SAMPLES,SAMPLE_BUFFERS,)

Create an OpenGL context before creating an ElasticFusion object, as ElasticFusion uses OpenGL internally. You can do this whatever way you wish, using Pangolin is probably easiest given it's a dependency:

    pangolin::Params windowParams;
    windowParams.Set("SAMPLE_BUFFERS", 0);
    windowParams.Set("SAMPLES", 0);
    pangolin::CreateWindowAndBind("Main", 1280, 800, windowParams);

I don't konw what the author means

wenjinshi, 2022-02-27:
ok,thank you a lot, I think maybe we shold do something as the readme file said in step 5 ,then we can run the dataset successfully,   Do you have any other social accounts， I want to keep in touch with you！

MichaelMano1, 2022-02-28:
@wenjinshi I have Line and Twitter account,do you use these ?

wenjinshi, 2022-02-28:
> @wenjinshi I have Line and Twitter account,do you use these ?

the second

MichaelMano1, 2022-02-28:
@wenjinshi  Can you try this way ,the Pangolin window may show up.
The error:
       Unrecognized options for scheme (SAMPLES,SAMPLE_BUFFERS,)
     
Delete  these two line in MainController.cpp:
 
      windowParams.Set("SAMPLE_BUFFERS", 0);
      windowParams.Set("SAMPLES", 0);

wenjinshi, 2022-02-28:
OK，thank you a lot.After this deletion , can your program run successfully?

wenjinshi, 2022-02-28:
and I found the code should be in file GUI.h ,not MainController.cpp  , isn't it?

wenjinshi, 2022-02-28:
I'm sorry to bother you for a long time. I have successfully debugged according to your method. Thank you

MichaelMano1, 2022-02-28:
@wenjinshi  Thank you as well.I still have some problem  with GPU on laptap,I will test it on desktop later.

MichaelMano1, 2022-02-28:
The error:
Unrecognized options for scheme (SAMPLES,SAMPLE_BUFFERS,)

Delete these two line in GUI.h:

  windowParams.Set("SAMPLE_BUFFERS", 0);
  windowParams.Set("SAMPLES", 0);

closed 2022-02-28 by MichaelMano1

wenjinshi, 2022-05-04:
Hello, excuse me. You helped me a lot before. Now I try to use other depth cameras to collect data. I successfully convert it into KLG format. Regardless of the size of the data, there will be segment errors and the core has been dumped. How should I solve this problem