mp3guy/ElasticFusion #235: GPUTest Error "Aborted (core dumped)"

Luoyingchao1, 2022-04-19:
Hello! Thanks in advance,I want to get the GPUConfig,so I run the GPUtest. However, I met a problem. The terminal tell me that "GPUTest: /home/nvidia/ElasticFusion/GPUTest/src/GPUTest.cpp:154: int main(int, char**): Assertion `argc == 2 && "Please supply the folder containing 1c.png, 1d.png, 2c.png and 2d.png"' failed." and "Aborted (core dumped)".
Can anyone know what happens? I run the ElasticFusion on TX2 and Ubuntu is 16.04.

hesom, 2022-04-28:
I believe either your current working directory has to be inside the folder with *.png files (the GPUTest folder) or you have to supply that folder as an argument to the GPUTest executable. For example if you are inside GPUTest/build you have to call `./GPUTest ..`

Luoyingchao1, 2022-04-28:
Thanks，this is helpful

closed 2022-05-27 by mp3guy

Luoyingchao1, 2025-04-15:
老师您好
毕业证、学位证和身份证扫描件已附在附件中
希望获取2018级自动化专业的录取花名册扫面件与个人成绩单（最大成绩即可）的扫描件，均需要带公章
感谢！！

LYC
***@***.***

&nbsp;

------------------ 原始邮件 ------------------
发件人:                                                                                                                        "湘潭大学档案馆"                                                                                    ***@***.***&gt;;
发送时间:&nbsp;2025年4月14日(星期一) 下午4:40
***@***.***&gt;;

主题:&nbsp;回复：2018级自动化专业本科录取名册和个人成绩单 中英文证明材料

您好，
1.查询本人学籍档案请同时提供本人身份证和毕业证（学位证）扫描件，并在邮件中注明所需材料和利用目的，如需邮寄请提供邮寄地址和联系电话。档案馆联系电话：0731-58293044。
2.我馆仅存有中文成绩单，如需英文成绩单请根据我馆保存的中文成绩单自行翻译（英文成绩单的顺序、内容、格式与中文成绩单一一对应）。
3.学历、学位证翻译件盖章请联系党政办。联系电话：0731-58292188。

&nbsp;
&nbsp;
------------------ 原始邮件 ------------------
***@***.***&gt;;
发送时间:&nbsp;2025年4月14日(星期一) 下午4:37
***@***.***&gt;; 

主题:&nbsp;2018级自动化专业本科录取名册和个人成绩单 中英文证明材料

&nbsp;

Hello
&nbsp; &nbsp; 老师好
&nbsp; &nbsp; 如刚刚电话沟通，我是咱们学校自动化专业本科毕业生，2018年入学，2022年毕业，
&nbsp; &nbsp; 姓名：罗应潮
&nbsp; &nbsp; 班级：自动化一班
&nbsp; &nbsp; 学号：201805551418
&nbsp; &nbsp; 现在因个人出国留学需要，应德国大使馆学历真实性审查要求（德国APS审查），需要收集个人和专业的相关材料（清单详细可见附件），麻烦档案馆老师能提供必要的便利，需要的资料如下：
&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 1.2018级自动化专业录取花名册中英文密封件
&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 2.本人毕业成绩单中英文密封件
&nbsp; &nbsp; &nbsp; &nbsp; &nbsp; 3.本人学位学历证明中英文密封件
&nbsp; &nbsp; &nbsp;
&nbsp; &nbsp; 其中，按照德国使馆要求，“中英双语密封件”是指由学校出具的、含中英文材料原件且加盖公章的密封信件。信封封口处要加盖 学校公章且密封。成绩单须由教务处或档案馆出具的“中英双语密封件”，仅有院系级盖章无效。

LYC
***@***.***

&nbsp;

qwert12131990, 2025-04-15:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。