mp3guy/ElasticFusion #242: Wins10 VS2019 CUDA10.1 build error

guan-lin, 2022-09-14:
Hi, professor. Thanks for your work. I am trying to study the Elastic Fusion on Wins. And i just compile this code on Windows10 with CMake 3.23 VS2019 CUDA10.1, and i get all the needed third-party, the CMake Configuring done and Generating done. but get error when i build the efusion lib:
**MSB8066  project_root/Core/Cuda/cudafuncs.cu; project_root/Core/Cuda/reduce.cu custom build has exited with code 1.**

How to solve it? Any advice will be helpful. Thank you very much.
Looking forward for your reply.

mp3guy, 2023-03-10:
Windows is no longer supported I'm afraid.

closed 2023-03-10 by mp3guy

zuriel921, 2023-06-07:
hello, i met the same problem, have you solve it?